oxidecomputer/transceiver-control #6: Naming request/response cases

jgallagher, 2022-11-11:
I noticed when @mkeeter pasted in an initial demo of his work talking to the SP that this struct: https://github.com/oxidecomputer/transceiver-control/blob/6139284e2e3c164db69b7b7e0be50ce133b49a28/messages/src/message.rs#L43-L50

uses the same naming style as MGS: https://github.com/oxidecomputer/management-gateway-service/blob/d7791e4d02571b8414b4c2976b0439234aa4d8c7/gateway-messages/src/lib.rs#L52-L60 but makes a different choice on meaning. In MGS, `MgsResponse` means "a response _from_ MGS" (i.e., sent by MGS in response to an `SpRequest`), but in transceiver-control `HostResponse` means "a response _to_ the host" (i.e., sent by the SP in response to a `HostRequest`). I think either is a reasonable interpretation but it might be surprising that we've chosen differently. I think at one point I had named the MGS cases `SpToHostResponse` to make it more clear; I'm also open to other names that would be more clear. :)

This is a very minor issue that doesn't affect the wire format at all (unless we reorder the enum), but wanted to note it here. I think it'd be useful to be more closely aligned in case folks are hopping between repos.

bnaecker, 2022-11-11:
Yeah, I noticed this as well. I agree both libraries should use the same naming scheme! One benefit of the existing MGS naming is that the peer portion always describes which peer _originated_ the message. It's probably clearer than what we've got here now, so I'm happy to move this over to that scheme at this point.

bnaecker, 2022-11-15:
Closed by #12