oxidecomputer/transceiver-control #8: Clamping `MemoryRegion` to 128 bytes?

mkeeter, 2022-11-14:
In SFF-8636, each page is only 128 bytes; if you start reading at address 100, you'll get 28 bytes of page 0, then will wrap around to the _beginning of page 0_ (instead of reading the upper page).

[image]

This means it's not actually possible to read the whole 256-byte region in one go.  Right now, `MemoryRegion` is checked to avoid reading outside of the 256-byte region; should we instead clamp it to avoid this kind of overflow?

rmustacc, 2022-11-14:
I would consider simplifying our lives such that you can read at most one page (which is the limit of the FPGA) and that we don't let higher level software wrap around unless we need it later. So in your example, we would return at most 28 bytes.

mkeeter, 2022-11-14:
Looking at CMIS, there's a similar wrap-around behavior.

[image]

In addition, there are limits to read and write sizes: by default you can only read / write 8 bytes in one go:

[image]

(SFF-8636 doesn't appear to have read length limitations, but you can only write 4 bytes in a single transaction)

I think it makes sense to have this stuff _checked_ on the SP, but probably not have the SP be responsible for automatically splitting up reads and writes?

rmustacc, 2022-11-14:
That seems reasonable to me at first blush.

bnaecker, 2022-11-14:
I think this all sounds fine, to limit the size of operations on the host as needed. For context on how we got where we are now, in SFF-8636, the wrap-around behavior is part of the auto-increment of the data word address counter However, one can use random reads/writes at a _specific address_. (See SFF-8636 rev 2.10a section 5.2.1 for the details on how that's accomplished.) It also appeared to me that [Hubris currently uses](https://github.com/oxidecomputer/hubris/blob/5729352bc444f89e60983b647586b23bc3f0f1b2/drv/sidecar-front-io/src/transceivers.rs#L342) the random read approach, meaning the wrap-around isn't relevant here, since the data word address is always written prior to the next operation anyway. So I believed it would be _possible_ to access the full memory-map in a single request to the SP, implying the SP would write to the data address word on each access.

But even if all the details about wrap-around are indeed the case, I didn't quite appreciate that Hubris will also fail requests to read more than [128 bytes](https://github.com/oxidecomputer/hubris/blob/a6b0055273c067af60c0c76abde5a54c096f362e/drv/transceivers-api/src/lib.rs#L40-L47), nor that the CMIS spec actually limits all accesses to 8 bytes. I think that all jives with the above, that we want to limit accesses to a single 128-byte section, and to 4- or 8-byte access, in the host.

That said, I'm fine implementing as much of the logic for splitting requests in the host. What I don't understand is whether we actually want or need to do that. What I mean is, during some initial discussions, it seemed like we needed some guarantees about atomicity in the protocol. It sounds like I've over-constrained that requirement. In particular, we need only atomic access to 4 bytes for SFF-8636 devices, and 8 bytes for CMIS -- is that correct?

The only other point I want to clarify is about accessing the upper pages. It still seems like one thing the SP must be responsible for is atomically writing the page-select byte, and then performing the requested client operation on it. I.e., we can limit accesses in the protocol to just the lower page or an upper page, but in that latter case, the SP has to perform a write to select it and then do the actual access. Does that all make sense?

mkeeter, 2022-11-14:
I don't see anything in SFF-8636 rev 2.10a section 5.2.1 about arbitrary-length reads.  I was going by 5.3.1 (Slave Memory Address Counter), and also experimentally: doing a 128-byte read starting at address 127 will give you page 0, shifted by 1!

Having the SP responsible for upper page management – and including that config in the same message as the actual read/write – still makes sense to me, so I don't think the data contents of `MemoryRegion` needs to change, just the constructor.

> In particular, we need only atomic access to 4 bytes for SFF-8636 devices, and 8 bytes for CMIS -- is that correct?

That seems correct to me for writes; reads are 128 bytes for SFF-8636 / 8 bytes for CMIS.  I'm not sure if we want to enforce that in the `MemoryRegion` constructor, but I'm going to at least add a check in the SP firmware.

bnaecker, 2022-11-14:
Sorry that was a typo, I meant to write section _5.3.1_, not 5.2.1. I also see where you get that writes are constrained to 4/8-bytes, but that reads for SFF-8636 are not:

> As long as the slave receives an acknowledge, it shall serially clock out sequential data words.

The experiment you ran is very helpful, thanks. I must have misunderstood what the `RandomRead` in the `setup_i2c_op` function meant. Perhaps it's only for the first read / write, and all subsequent accesses are sequential.

>  I'm not sure if we want to enforce that in the MemoryRegion constructor, but I'm going to at least add a check in the SP firmware.

I'd like to at least try to encode all this information in the type system on the host as much as possible, if that seems reasonable to you. I will experiment with read/write-specific memory region types, but abandon if it gets too verbose or annoying.

mkeeter, 2022-11-14:
👍🏻 just tag me for review when you're feeling good about it!

Aaron-Hartwig, 2022-11-15:
Likewise!

bnaecker, 2022-11-15:
Completed by #11

closed 2022-11-15 by bnaecker