oxidecomputer/transceiver-control #17: Can read from transceiver modules that are not present

bnaecker, 2022-11-18:
This could be a bug here or in the Hubris side of the transceiver management, I'm not yet sure. I'm recording my state here before we forget. cc @mkeeter @Aaron-Hartwig 

In fleshing out `xcvradm` and the `Controller` library, I added a subcommand to get the status of all modules, and read the lower memory page. I'm currently just reading the first byte of the lower page, which is always an SFF-8024 style identifier for modules that conform to either SFF-8636 or CMIS management specs. This is to start building out the code for parsing memory maps in the host libraries.

While doing that, I noticed that it seems possible to read from a module that does _not_ have the `PresenceL` pin asserted:

```
ben@niles ~ $ ./xcvradm -i axf0 status
FPGA    Port    Status
0       0       PRESENT | ENABLED | LOW_POWER_MODE | INTERRUPT
0       1       PRESENT | ENABLED | LOW_POWER_MODE | INTERRUPT
0       2       RESET | LOW_POWER_MODE
0       3       RESET | LOW_POWER_MODE
0       4       RESET | LOW_POWER_MODE
0       5       RESET | LOW_POWER_MODE
0       6       RESET | LOW_POWER_MODE
0       7       RESET | LOW_POWER_MODE
0       8       RESET | LOW_POWER_MODE
0       9       RESET | LOW_POWER_MODE
0       10      RESET | LOW_POWER_MODE
0       11      RESET | LOW_POWER_MODE
0       12      RESET | LOW_POWER_MODE
0       13      RESET | LOW_POWER_MODE
0       14      RESET | LOW_POWER_MODE
0       15      RESET | LOW_POWER_MODE
```

So that appears to say we only have a module plugged in to the first 2 ports of the FPGA, and the rest are missing. And yet:

```
ben@niles ~ $ ./xcvradm -i axf0 -t 3 read-lower 0 4
FPGA    Port    Data
0       3       [0x00,0x00,0x00,0x00]
```

I seem to be able to read from port 3, even though the presence bit is missing from the status bitmap. Reads from the first two ports seem OK:

```
ben@niles ~ $ ./xcvradm -i axf0 -t 0,1 read-lower 0 4
FPGA    Port    Data
0       0       [0x11,0x07,0x00,0x2f]
0       1       [0x11,0x08,0x00,0x0f]
```

Those seem reasonable, in that `0x11` is the SFF-8024 identifier for a QSFP-28 module, which I believe these are. It actually turns out I can read from _every_ transceiver expect port 2:

```
ben@niles ~ $ for i in $(seq 0 15); do ./xcvradm -i axf0 -t $i read-lower 0 4; done
FPGA    Port    Data
0       0       [0x11,0x07,0x00,0x2f]
FPGA    Port    Data
0       1       [0x11,0x08,0x00,0x0f]
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: ReadFailed', controller/src/bin/xcvradm.rs:141:61
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
FPGA    Port    Data
0       3       [0x00,0x00,0x00,0x00]
FPGA    Port    Data
0       4       [0x00,0x00,0x00,0x00]
FPGA    Port    Data
0       5       [0x00,0x00,0x00,0x00]
FPGA    Port    Data
0       6       [0x00,0x00,0x00,0x00]
FPGA    Port    Data
0       7       [0x00,0x00,0x00,0x00]
FPGA    Port    Data
0       8       [0x00,0x00,0x00,0x00]
FPGA    Port    Data
0       9       [0x00,0x00,0x00,0x00]
FPGA    Port    Data
0       10      [0x00,0x00,0x00,0x00]
FPGA    Port    Data
0       11      [0x00,0x00,0x00,0x00]
FPGA    Port    Data
0       12      [0x00,0x00,0x00,0x00]
FPGA    Port    Data
0       13      [0x00,0x00,0x00,0x00]
FPGA    Port    Data
0       14      [0x00,0x00,0x00,0x00]
FPGA    Port    Data
0       15      [0x00,0x00,0x00,0x00]
ben@niles ~ $
```

Note for those wanting to use `xcvradm`: You can do reads of all the ports in one go with `xcvradm read-lower 0-4` (omitting the `-t` flag), but that will fail with a `ReadFailed` error. That's because reading from port 2 fails, which fails the whole request.

Aaron-Hartwig, 2022-11-18:
This is bug, but _where_ the bug it is may be up for debate 😅. We aren't screening I2C transaction attempts at the SP or the FPGA, so we attempt to talk to transceivers which are not there. This results in a NACK, but unless the SP looks at the FPGA's error register, it won't know this happened. It will just look at the busy register, see the transaction has finished, and then will go read data from the read buffer for that port (which will be stale!). So the fix here is to prevent I2C reads/writes to modules which are not in A2.

Should host software prevent this from happening, or do we want the SP to? My vote would be the host since that would prevent the communications at the source.

bnaecker, 2022-11-18:
I'm not sure I understand the full data path from module -> FPGA -> SP. It _seems_ like:

- The FPGA does an I2C transaction to read some data. (I gather it can do that regardless of whether there's an I2C follower / slave? Or maybe that there's always a buffer for the SP to pull from, see below.)
- If the transaction fails, the FPGA set a bit somewhere that the SP can query.
- The SP currently polls the transaction-completed register to see when the I2C transaction finishes. It _could_ conceivably check the error register IIUC.
- When that finishes, the SP reads from a buffer that the FPGA _would have_ put data into, assuming the transaction completed successfully. This is another read that seems possible regardless of whether the actual module is present, I think? That is, the FPGA always has a buffer the SP can read from, but that data may be stale or meaningless or otherwise undefined.

Is that all accurate?

Another question I have is why this only seems to generate an error on port 2?

In any case, I totally agree with the general disposition that the host should do as much work as possible. That makes a lot of sense. Still, I don't see how that would be possible in this case, at least without a race. Suppose the following sequence happens:

- The host asks the Hubris `transceivers` for the modules that are present, and gets back some bitmask.
- One of those modules is unplugged.
- The host sends a new read operation, addressing only those modules in the mask, and gets back data.

It has no way of knowing that the data for that now-unplugged module is invalid or wrong, and doing an additional check at the end isn't enough to verify that.

The SP may also be susceptible to this race, to be sure. I don't know enough about the FPGA's logic and how I2C works to know if that's _always_ going to be true. But it's possible we could at least _minimize_ this race window by having the SP AND the bitmask it gets from the host with the current presence bitmask. It could send back a new bitmask in the response to the host, which it would be obliged to check to before assuming anything about the data.

I dunno what the right call here is. The host can do some checks itself, for sure. And I also think we can guarantee that at some point, we're going to try to read / write a yanked module. I guess the issue is (1) Does that matter? and (2) how can we tell?

rmustacc, 2022-11-18:
This is a case where the host can't do this for all the reasons that Ben mentioned. It also applies to the SP. Fundamentally only the initiator (the FPGA) can really tell this has happened without a race. So unfortunately @Aaron-Hartwig I think this is something that we need to actually implement in the FPGA. Ideally what we would see is:

* When the SP directs a read or write to the FPGA and there is no device present, we get a crisp failure mode about that (note it could also be removed while the operation is ongoing) recorded in the per-port transaction results.
* The SP looks at that and includes that information back in its reply to the host, not sending data across for that port.
* The host looks at the specific semantic error and tries to figure out what the right next step is.

So today we should get a failed transaction note at least and that has to make it out from FPGA->SP->Host. Ideally we would make those error codes more detailed with more info as the first question when someone gets a classic generic thing failed error is to ask why. We can always start with generic, this failed, dunno error and work out and add in more detail.

Aaron-Hartwig, 2022-11-18:
>Is that all accurate?

Your understanding is correct @bnaecker.

>Another question I have is why this only seems to generate an error on port 2?

That is unexpected for me as well.

Points well taken on the race condition here. While I would prefer that the Host layer not initiate transactions on modules it does not think are there, we definitely need to handle the case where a module is removed but that state has not propagated to the Host yet. We have a status register at the SP checks which includes `BUSY` and `ERROR` flags, as well as an `ERROR_TYPE` field. Attempting to initiate an I2C transaction on a module that is not present can be treated as an error condition, setting the `ERROR` flag and getting its own custom `ERROR_TYPE`.

mkeeter, 2022-11-18:
For what it's worth, I'm _also_ confused where the bug lies: the SP code should be checking the error register after I2C transactions, which I'd expect to be set if a transceiver isn't present (because it won't NAK).

mkeeter, 2022-11-18:
Okay, I'm now suspicious that this is an FPGA-level bug, where it's not reporting NACKs.

Trying to read from port 3 (which is presumably unplugged), I never see the error bit set in `QSFP_PORT3_I2C_STATUS`
```
humility: ring buffer drv_transceivers_server::net_api::__RINGBUF in transceivers:
 NDX LINE      GEN    COUNT PAYLOAD
   0  285        1        1 Read(ModuleId { fpga_id: 0x0, ports: PortMask(0x8) }, MemoryRead { page: Sff8636(Lower), offset: 0x0, len: 0x4 })
   1  433        1        1 GotStatus(0x3, 0x10)
   2  433        1        1 GotStatus(0x3, 0x0)
```
I read the status register twice.  Once, it replies with `BUSY (0x10)` set, then replies without any bits set.

Contrast this with port 2, which correctly returns the error bit set:
```
   3  285        1        1 Read(ModuleId { fpga_id: 0x0, ports: PortMask(0x4) }, MemoryRead { page: Sff8636(Lower), offset: 0x0, len: 0x4 })
   4  433        1        1 GotStatus(0x2, 0x8)
```

Aaron-Hartwig, 2022-11-18:
Yeah, this is certainly a board-level issue. I'm working on understanding if it is a design flaw or a problem introduced by the partial-board assembly. I just attempted to read from ports 0-3 with all the modules disabled, this means that there is no 3.3V at each module, meaning the I2C bus will not even have its pull-ups. Yet, I see port 2's bus being active and the SCL on port 3 being pulled up. 

[image]

The buses being active on port 2 and SCL of port 3 make no sense, so I'm going to probe around.

Aaron-Hartwig, 2022-11-18:
So after looking at the schematics I noticed (remembered?) I had added bleeder resistors onto the eFuse rails since they leak significantly when `EN` is low. So the problem of the bus floating will be resolved on Build B hardware (details in https://github.com/oxidecomputer/hardware-qsfp-x32/issues/47). 

In the short term, I'm going to have the FPGA not even attempt I2C if we don't have a module both present and enabled. Longer term this will be modeled by not allowing I2C access if the module is not in A2 power state.

bnaecker, 2022-11-18:
It occurs to me that we need to take similar steps when doing things like transitioning power modes. That may happen at the SP rather than the FPGA, though, I'm not sure. But the host will certainly only _try_ to enable power for modules that are present, but the SP or FPGA would need to AND that with the currently present modules I think.

mkeeter, 2022-11-18:
Agreed; that would would be best done on the FPGA side, since the FPGA will be both in charge of power states _and_ ground truth for transceiver presence.

Aaron-Hartwig, 2022-11-22:
This will be resolved by https://github.com/oxidecomputer/hubris/pull/950.

closed 2022-11-22 by bnaecker