oxidecomputer/transceiver-control #27: Upper page data is identical regardless of actual page selected

bnaecker, 2022-11-28:
It appears that we always receive _identical_ data when reading a module's upper memory page, regardless of the actual page we request. This occurs using both `xcvradm` and @mkeeter's demo CLI, so appears to be an issue either on the SP or the FPGA. I'll show the bug here using `xcvradm`.

```bash
ben@niles ~ $ ./xcvradm -i axf0 status
FPGA Port Status
   0    0 PRESENT | ENABLED | LOW_POWER_MODE | INTERRUPT
   0    1 PRESENT | ENABLED | LOW_POWER_MODE | INTERRUPT
   0    2 RESET | LOW_POWER_MODE
   0    3 RESET | LOW_POWER_MODE
   0    4 RESET | LOW_POWER_MODE
   0    5 RESET | LOW_POWER_MODE
   0    6 RESET | LOW_POWER_MODE
   0    7 RESET | LOW_POWER_MODE
   0    8 RESET | LOW_POWER_MODE
   0    9 RESET | LOW_POWER_MODE
   0   10 RESET | LOW_POWER_MODE
   0   11 RESET | LOW_POWER_MODE
   0   12 RESET | LOW_POWER_MODE
   0   13 RESET | LOW_POWER_MODE
   0   14 RESET | LOW_POWER_MODE
   0   15 RESET | LOW_POWER_MODE
ben@niles ~ $ ./xcvradm -i axf0 -t0,1 identify
FPGA Port Identifier       Vendor           Part             Rev  Serial           Mfg date
   0    0 Qsfp28 (0x11)    Intel Corp       SPTSBP3CLCCO     01   CRCL203602EHS    14 Sep 2020 (Lot 00)
   0    1 Qsfp28 (0x11)    FS               QSFP28-SR4-100G  1A   F2220590150      15 Jun 2022
ben@niles ~ $ ./xcvradm -i axf0 -t 0 read-upper --sff --page 0 128 4
FPGA Port Data
   0    0 [0x11,0xcc,0x07,0x80]
ben@niles ~ $ ./xcvradm -i axf0 -t 0 read-upper --sff --page 2 128 4
FPGA Port Data
   0    0 [0x11,0xcc,0x07,0x80]
ben@niles ~ $ ./xcvradm -i axf0 -t 0 read-upper --sff --page 3 128 4
FPGA Port Data
   0    0 [0x11,0xcc,0x07,0x80]
```

So that shows that the first four bytes of the upper page are identical, regardless of the page that we actually request. (Note that the same is true of _all_ 128 bytes of the page, I'm only showing the first 4 for clarity.) The Hubris `transceivers` task should be doing page-selection here, but it doesn't appear that the page-select byte is updated:

```bash
ben@niles ~ $ ./xcvradm -i axf0 -t 0 read-lower --sff 127 1
FPGA Port Data
   0    0 [0x00]
```

Note that there's a few causes here. First, the SFF-8636 spec requires that free-side modules revert to upper page 0 when the fixed-side requests a page that's not supported. However, I don't _think_ that's what's going on here. We can first see that the module supports paged memory:

```bash
ben@niles ~ $ ./xcvradm -i axf0 -t 0 read-lower --sff 2 1 --binary
FPGA Port Data
   0    0 [0b00000000]
```

Bit 2 is zero, which means memory is paged for this module. So we can read at least upper page 0 and 3. Upper page 0, byte 195 includes details of which other upper pages are supported as well:

```bash
ben@niles ~ $ ./xcvradm -i axf0 -t 0 read-upper --sff 195 1 --binary
FPGA Port Data
   0    0 [0b10111110]
```

From SFF-8636 Table 6-21, bit 7 (which is 1 here) advertises that the free-side implements page 2, which is a user-writable memory page. So, we should certainly be able to access upper pages 0, 2, and 3. (Pages 4 and above are vendor-specific, and I've not checked if this part actually implements them.)

I've also tried to explicitly _set_ the page-select byte myself, and then read the data back. That doesn't seem to work, even though page 3 should certainly be implemented:

```bash
ben@niles ~ $ echo 3 | ./xcvradm -i axf0 -t 0 write-lower --sff --input-kind hex 127  # Write 0x03 to the page select byte
ben@niles ~ $ ./xcvradm -i axf0 -t 0 read-lower --sff 127 1  # Read the 1 byte starting at the page-select offset
FPGA Port Data
   0    0 [0x00]
```

cc @mkeeter @Aaron-Hartwig

mkeeter, 2022-11-28:
Instrumenting the Hubris side, we're definitely trying to select the page (i.e. decoding the message correctly), and I don't think this part of the code has changed much.

Using the `demo` app (i.e. a totally different host codebase), I see byte 127 **not changing** when we attempt to select the page.  This is the same as the observations above, and is definitely suspicious.

Aaron is investigating the FPGA side now!

Aaron-Hartwig, 2022-11-28:
This is definitely an FPGA regression. I am going through steps to reproduce it locally and am going to dust off and improve an aged set of module control simulations that could catch something like this.

Aaron-Hartwig, 2022-11-29:
@bnaecker please take a look at https://github.com/oxidecomputer/hubris/pull/963 and see if that resolves the issue.

bnaecker, 2022-11-29:
@Aaron-Hartwig flashed the Hubris archive with the fix above to `niles`, and I've confirmed that I can successfully (1) set the page-select byte manually and read it back, and (2) that basic write support works, by writing/reading page 0x02 (the user-reserved scratch page).

```
ben@niles ~ $ ./xcvradm -i axf0 -t0 status
FPGA Port Status
   0    0 PRESENT | ENABLED | LOW_POWER_MODE | INTERRUPT
ben@niles ~ $ ./xcvradm -i axf0 -t0 identify
FPGA Port Identifier       Vendor           Part             Rev  Serial           Mfg date
   0    0 Qsfp28 (0x11)    Intel Corp       SPTSBP3CLCCO     01   CRCL203602EHS    14 Sep 2020 (Lot 00)
ben@niles ~ $ ./xcvradm -i axf0 -t0 read-lower --sff 127 1
FPGA Port Data
   0    0 [0x00]
ben@niles ~ $ echo '2' | ./xcvradm -i axf0 -t0 write-lower --sff --input-kind hex 127
ben@niles ~ $ ./xcvradm -i axf0 -t0 read-lower --sff 127 1
FPGA Port Data
   0    0 [0x02]
ben@niles ~ $ ./xcvradm -i axf0 -t0 read-upper --sff --page 2 128 4
FPGA Port Data
   0    0 [0x05,0x06,0x07,0x08]
ben@niles ~ $ echo '1 2 3 4' | ./xcvradm -i axf0 -t0 write-upper --sff --page 2 128 --input-kind hex
ben@niles ~ $ ./xcvradm -i axf0 -t0 read-upper --sff --page 2 128 4
FPGA Port Data
   0    0 [0x01,0x02,0x03,0x04]
```

This issue seems resolved by https://github.com/oxidecomputer/hubris/pull/963!

closed 2022-11-29 by bnaecker