oxidecomputer/transceiver-control #29: Want support for more CMIS advertised pages

bnaecker, 2022-11-30:
#26 added basic support for decoding the pages a module advertises support for. SFF-8636 modules are pretty simple here, but CMIS is a good deal more complicated. We're parsing a few of the more straightforward pages, but [here is a list](https://github.com/oxidecomputer/transceiver-control/blob/f00655f134fbfa6eddb95ac81e5dfdbb9d2696b1/decode/src/lib.rs#L527) of some others we should add support for. That requires reading a few more bytes from the module and decoding the right bits. Not complicated, but not critical at this point.

Aaron-Hartwig, 2023-07-12:
I would like to add the following to the wishlist:
>Read page 0x01 byte 142, where bit 5 indicates support for page 0x13 and 0x14, Module Performance Diagnostics Control and Results.

bnaecker, 2023-07-12:
Yeah, that'd be nice. During the racktest whirlwind, we ended up reading lots of data out of those pages, such as the LOS / LOL flags. It'd be very nice to have a high-level decoding command that extracts whatever information is available about the physical layer, things like those LOS flags.