oxidecomputer/transceiver-control #36: Need API to specifically assert and deassert `ResetL` line

bnaecker, 2022-12-08:
We currently have a message that allows the host to [reset a module](https://github.com/oxidecomputer/transceiver-control/blob/5b13d464f2b62a058a2c678f59856bc9f3c963a9/messages/src/message.rs#L102). That has the semantics of first asserting and then immediately deasserting the `ResetL` pin on the actual QSFP transceiver modules. That's indeed how it's [implemented in Hubris](https://github.com/oxidecomputer/hubris/blob/de7848c4177f3c4f5e56e204fee35d7b00c34afa/drv/transceivers-server/src/udp.rs#L332-L343). That works OK for now, because the sleep time in between those operations is 1ms, which is longer than the specification mandates to actually trigger module reset (10us).

But when bringing up the front IO board on a Sidecar using the Tofino ASIC, it became clear that the BF SDE drivers really expect to have full control to drive that pin whichever way they want for an arbitrary period of time. The messaging API does not support that.

One way to do this would be to add a trailing `u8` to the `Reset` message, with a 0 meaning "deassert `ResetL`" and a 1 meaning "assert `ResetL`". I'm sure there are other ways, too, so open to discussion there. CC @Aaron-Hartwig @mkeeter

Aaron-Hartwig, 2022-12-08:
This will be a small FPGA adjustment and a tiny SP change. I'll track that by https://github.com/oxidecomputer/quartz/issues/67

mkeeter, 2022-12-08:
I have the opposite preference – I'd vote to make the behavior obvious in the variant name (so `AssertReset` / `DeassertReset`, or a combined `SetResetPin(bool)`), rather than having secret special behavior if there's a payload.

bnaecker, 2022-12-08:
@mkeeter That sounds like a fine approach to me too. Of those, I have a slight preference for the data-carrying enum. @Aaron-Hartwig thoughts on those? I can plug this in today, if we're in agreement. Speaking of which, I think we need to start pinning the controller and Hubris software to a rev, so we can update these things on their own cadence. We could tag the commits for each version, maybe, and then specify that in the Cargo.toml? I've never done that before, but as long as we're not on crates.io it might be useful.

mkeeter, 2022-12-08:
In Hubris, we typically rely on the revision being stored in `Cargo.lock` (which is committed to the repo); we can update Git dependencies with `cargo update -ptransceiver-messages` when we want to bump them.

Aaron-Hartwig, 2023-01-04:
Wow, sorry I did not follow up on this convo in a timely manner. I don't have a strong preference here. Given this is a binary operation (asserting or de-asserting `ResetL`), I'm fine with taking the "data" out of the payload and making the action easily viewable via an explicit message type for each action. I'm touching all this stuff right now so I can implement this and have y'all take a peek when I'm done.

Aaron-Hartwig, 2023-01-27:
Implemented by https://github.com/oxidecomputer/transceiver-control/pull/54

closed 2023-01-27 by Aaron-Hartwig