oxidecomputer/transceiver-control #41: Host needs to check for software control of `LPMode`

bnaecker, 2022-12-08:
The BF SDE appears to quickly set byte 93, bit 1 of SFF-8636 modules. That is used by host software to inform the module that the `LPMode` pin should be ignored, and that its functionality is subsumed by byte 93, bit 1. In this case, reading the status back in the host may show `LOW_POWER_MODE` in the status bits, but that's not the whole story. The _actual_ power mode may need to be read from the memory map itself.

This issue tracks putting those smarts into the host, in something like a `get_power_state` command. That looks for software control, and then returns the disposition of the `LPMode` pin or byte 93, bit 2. Note that we may also want to return more power information here, such as the power class.

bnaecker, 2023-01-24:
Note that this also affects the way we _set_ the power mode! Currently, the `SetPowerMode` command sent from host to SP will tell the SP to assert or de-assert the `LPMode` pin. That's only sufficient if the module isn't currently set to software control of the power state. We need to check for that, then either set LPMode or write the correct value to the module's memory map depending on its kind.

bnaecker, 2023-01-25:
I'm prototyping this and starting to think it's a bigger change. In particular, I don't believe our current `SetPowerMode` message is a great interface for this. I think we need a message to specifically enable or disable power entirely (the efuse), and also to "pass-through" the `LPMode` signal, the same way we're planning to pass `ResetL`.

The reason is basically that the three `PowerMode`s "work" pretty differently. Turning things off can always be done, regardless of the module. We control that power rail independent of everything else. Turning things _on_, either into low or high-power, works differently. If the module is under hardware control, then we twiddle the `LPMode` pin. If it's under hardware control, we need to write to a specific bit, depending on the kind of module.

That's all fine, but it kind of obviates the purpose of the single `PowerMode` enum in our message. We could still use it in the `Controller` API, but it might be better to have:

- A message for turning on or off the power rails: `SetPowerOn(bool)` as a strawman.
- A message for twiddling `LPMode`: `SetLpActive(bool)` as a strawman.
- We can use the existing memory-write messages for the other case.

All of this is inline with generally keeping the smarts in the host. @Aaron-Hartwig I know you're working on something similar with `ResetL`, does this intersect with that at all? @mkeeter Any thoughts?

To be clear, the existing messaging API will _work_, since we can just not use the enum variants `PowerMode::{Low,High}` if the module is configured for software override of power. It just feels awkward and possibly confusing.

Aaron-Hartwig, 2023-01-25:
@bnaecker I think I provide everything we need here with: https://github.com/oxidecomputer/transceiver-control/pull/54

It provides the path we _want_ from the control plane using the `PowerMode` enum, while also supporting the path we _need_ currently which needs direct control of the `ResetL` and `LpMode`.

bnaecker, 2023-02-01:
Closed by #55

closed 2023-02-01 by bnaecker