oxidecomputer/transceiver-control #59: Need better handling of version mismatches

bnaecker, 2023-02-09:
While getting rack2 stood up, @jclulow encountered issues bringing up one of the front IO QSFP ports. The crux of the problem was a version mismatch between the host and SP. This was obscured for several reasons, and suggest that we generally need much better handling of version mismatches and some amount of backward-compatibility.

The transceiver was correctly powered with `xcvradm -i sidecar0 -t0 assert-lpmode`, which we can verify:

```bash
BRM42220051-switch # xcvradm -i sidecar0 -t0 status
FPGA Port Status
   0    0 PRESENT | ENABLED | LOW_POWER_MODE | POWER_GOOD
BRM42220051-switch # xcvradm -i sidecar0 -t0 power
FPGA  Port  Power  Software-override
   0     0  Low    No
BRM42220051-switch # xcvradm -i sidecar0 -t0 vendor-info
FPGA Port Identifier           Vendor           Part             Rev  Serial           Mfg date
   0    0 QsfpPlusSff8636 (0x0d) Aris             CAB-Q-Q-5M       01   Q210606290004    02 Jul 2021
BRM42220051-switch #
```

So the SP can be reached via `sidecar0` from the switch zone. However, we see this in the Dendrite logs:

```bash
BRM42220051-switch # tail -f $(svcs -L dendrite) | grep ERRO
Dec 28 04:23:01.481 ERRO 1986-12-28 04:23:01.481249 BF_PLTFM ERROR - PM_INTF QSFP Error Communication Failed with Hardware: in scanning qsfps at qsfp_scan_timer_cb:479
Dec 28 04:23:01.481 ERRO failed to detect front IO module presence, reason: Transceiver(MaxRetries(3)), unit: qsfp-ffi
Dec 28 04:23:03.481 1986-12-28 04:23:03.481191 BF_PLTFM ERROR - PM_INTF QSFP Error Communication Failed with Hardware: in scanning qsfps at qsfp_scan_timer_cb:479
ERRO failed to send message within 3 retries, task: io, unit: transceiver-controller
Dec 28 04:23:03.481 ERRO failed to detect front IO module presence, reason: Transceiver(MaxRetries(3)), unit: qsfp-ffi
Dec 28 04:23:05.482 ERRO failed to send message within 3 retries1986-12-28 04:23:05.482456 BF_PLTFM ERROR - PM_INTF QSFP Error Communication Failed with Hardware: in scanning qsfps at qsfp_scan_timer_cb:479
Dec 28 04:23:05.482 ERRO failed to detect front IO module presence, reason: Transceiver(MaxRetries(3)), unit: qsfp-ffi
Dec 28 04:23:07.481 1986-12-28 04:23:07.481750 BF_PLTFM ERROR - PM_INTF QSFP Error Communication Failed with Hardware: in scanning qsfps at qsfp_scan_timer_cb:479
ERRO failed to send message within 3 retries, task: io, unit: transceiver-controller
```

That shows that `dpd` isn't getting any responses. It is definitely sending the messages though, which we can verify with `snoop`:

```bash
BRM42220051-switch # snoop -r -d sidecar0
Using device sidecar0 (promiscuous mode)
fe80::aa40:25ff:fe05:0 -> ff02::1      UDP D=8888 S=997 LEN=22
fe80::aa40:25ff:fe05:0 -> ff02::1      UDP D=8888 S=997 LEN=22
fe80::aa40:25ff:feb6:c06c -> ff02::1de:2  UDP D=11112 S=11112 LEN=22
fe80::aa40:25ff:feb6:c06c -> ff02::1de:2  UDP D=11112 S=11112 LEN=22
fe80::aa40:25ff:feb6:c06c -> ff02::1de:2  UDP D=11112 S=11112 LEN=22
```

Those last three are the attempts to deliver the message to the SP. Now the most common cause is a protocol version mismatch. We can't directly verify this in the normal approach however, looking at `humility ringbuf`, because there are no debug probes in this system. The hot-off-the-presses dump-over-IP support in `humility` is _critical_ here. I was able to take a core file and run `humility ringbuf` against that, which indeed diagnoses the issue:

```bash
ben@jeeves ~ $ pfexec humility --archive /net/catacomb/data/staff/dock/rack2/sidecar/b/sp/build-sidecar-b.zip --ip fe80::aa40:25ff:fe05:401%rack2sw0tp0 dump
humility: connecting to fe80::aa40:25ff:fe05:401%rack2sw0tp0
humility: initializing dump agent state
humility: initializing segments
humility: taking dump; target will be stopped for ~20 seconds
humility: pulled 156.59KB in 0 seconds
humility: dumping to hubris.core.0
humility: dumped 764.92KB in 20 seconds
ben@jeeves ~ $ ls
README         bin            hubris.core.0  sc             stlouis        xcvradm
ben@jeeves ~ $ humility -d hubris.core.0 ringbuf transceivers
humility: attached to dump
humility: ring buffer drv_transceivers_server::__RINGBUF in transceivers:
 NDX LINE      GEN    COUNT PAYLOAD
   0  584        1        2 FrontIOReady(false)
   1  574        1        1 FrontIOReady(true)
   2  612        1        1 LEDInit
   3  140        1        1 LEDInitComplete
humility: ring buffer drv_transceivers_server::udp::__RINGBUF in transceivers:
 NDX LINE      GEN    COUNT PAYLOAD
   0  263        2        1 Status(ModuleId { fpga_id: 0x0, ports: PortMask(0xffff) })
   1  201        2        3 WrongVersion(0x3)
   2  263        2        1 Status(ModuleId { fpga_id: 0x1, ports: PortMask(0xffff) })
   3  201        2    51561 WrongVersion(0x3)
   4  263        2        1 Status(ModuleId { fpga_id: 0x0, ports: PortMask(0xffff) })
   5  201        2        3 WrongVersion(0x3)
   6  263        2        1 Status(ModuleId { fpga_id: 0x1, ports: PortMask(0xffff) })
   7  201        2        9 WrongVersion(0x3)
   8  301        2        1 AssertLpMode(ModuleId { fpga_id: 0x0, ports: PortMask(0xffff) })
   9  201        2        6 WrongVersion(0x3)
  10  301        2        1 AssertLpMode(ModuleId { fpga_id: 0x1, ports: PortMask(0xffff) })
  11  201        2        3 WrongVersion(0x3)
  12  263        2        1 Status(ModuleId { fpga_id: 0x1, ports: PortMask(0xffff) })
  13  201        2        6 WrongVersion(0x3)
  14  263        2        1 Status(ModuleId { fpga_id: 0x0, ports: PortMask(0xffff) })
  15  201        2    41042 WrongVersion(0x3)
ben@jeeves ~ $
```

So that's 41K `WrongVersion` messages that the SP logged. We need better surfacing of this failure mode, for one thing. Another is that we need to commit to some amount of cross-version compatibility at this point. The SP and host _will_ be updated separately, and we need to assume some of those updates will include changes to this protocol.

mkeeter, 2023-02-14:
Because of how Hubpack handles serialization, adding new variants to the _end_ of the `enum` is only a _semi_-breaking change, so we may not want to bump the version in that case:

- Old messages will be encoded / decoded as usual
- The new variants will return a Hubpack deserialization error if they're sent to an older firmware

rmustacc, 2023-02-14:
cc @smklein as we're tracking versions / upgrades we're going to need to be keenly aware that the SP version on Sidecar and switch zones need to stay in sync. This isn't unique, but wanted to highlight an additional case.

bnaecker, 2023-02-14:
@mkeeter That's a good point that adding variants at the end doesn't break handling of existing variants. I'm not positive how it works with variants that contain data, such as [this one](https://github.com/oxidecomputer/transceiver-control/blob/cec7bd168b7a447a28db0fd4b2910a8d3ec63e01/messages/src/lib.rs#L92). But assuming the same rules apply to the _inner_ enum as well, then that would mean all variants added to the end _recursively_ would bump the version number, but not break decoding of existing variants.

I've been prototyping something to handle this. My current plan is to add a min / max version that are supported. The max is by definition the current version. When we add new variants, we'd bump the current / max version, but not change the minimum. When we reorder variants at any point, we'd need to bump the min up to the current. To be clear, I think we should see if we can commit to basically _never_ doing that at this point. Most critically, it would help to make sure we never reorder the existing variants with respect to those needed to communicate a deserialization failure or version mismatch.

That would mean the following could not be reordered:

- Any variant of [`MessageBody`](https://github.com/oxidecomputer/transceiver-control/blob/main/messages/src/message.rs#L71). This should be fine.
- [`SpResponse::Error`](https://github.com/oxidecomputer/transceiver-control/blob/main/messages/src/message.rs#L140)
- [`HostResponse::Error`](https://github.com/oxidecomputer/transceiver-control/blob/main/messages/src/message.rs#L181)
- The [`Error` variants](https://github.com/oxidecomputer/transceiver-control/blob/cec7bd168b7a447a28db0fd4b2910a8d3ec63e01/messages/src/lib.rs#L104-L134) responsible for indicating a protocol or version error

If that all sounds correct, I also plan to write tests to assert this. Those would basically take the existing variant indices, which is how `hubpack` [seems to be generating the serialization](https://docs.rs/hubpack_derive/0.1.1/src/hubpack_derive/lib.rs.html#43), and test that they serialize/deserialize to the expected variants of the above. This will at lest let us know when we need to bump the _minimum_ version number, by definition up to the current. It's not perfect, but better than the current system. Let me know what you think, if I've missed something or there's a better approach.

bnaecker, 2023-02-28:
Addressed for now by #66

closed 2023-02-28 by bnaecker