oxidecomputer/transceiver-control #63: Add derives for using some data in Dropshot APIs

bnaecker, 2023-02-24:
There are a few types that we'd like to use in Dropshot APIs. This will be at least Dendrite, but some of these may also be exported all the way to the internal or even external Nexus API as well. We should add a feature (with associated optional dependencies) for deriving `serde::{Deserialize, Serialize}` and `schemars::JsonSchema` for the following types:

- [ ] `PowerMode`: This is the primary way folks will view or control the power of a transceiver
- [ ] `VendorInfo`: The primary identifying information for a transceiver. It'll be most notably used to derive topology information and unique identifiers for each transceiver. See [RFD 360](https://rfd.shared.oxide.computer/rfd/0360#_qsfp_transceivers).
- [ ] Health and monitoring data: These don;t exist yet in `transceiver-decode`, but will need to soon, and will need to be visible via Dendrite's API. This includes: (1) current / voltage; (2) per-lane optical power; (3) cage temperature; (4) possibly alarm or fault state

It's not clear what we should do with `Status`. I'm inclined to think this should be changed from a bitmap to a struct, showing the information per-transceiver e.g., as boolean fields.

Aaron-Hartwig, 2023-02-24:
>It's not clear what we should do with Status. I'm inclined to think this should be changed from a bitmap to a struct, showing the information per-transceiver e.g., as boolean fields.

That seems logical to me. What would keeping it as a bitmap get us?

bnaecker, 2023-02-24:
I dunno. I was wondering if we would serialize it as a string, which I think `bitflags` does like `FOO | BAR | BAZ`. That's not super helpful, so a struct seems the way to go. I'll play around with it.

bnaecker, 2023-05-15:
Completed by #68

closed 2023-05-15 by bnaecker