oxidecomputer/transceiver-control #105: `xcvradm` could use parseable output

bnaecker, 2023-06-21:
Many illumos tools support a parseable mode, in which the output is regularly-structured and meant for machine consumption. Rather than nicely-printed tables, output is formatted with a single, often configurable, separator character. `xcvradm` could benefit from such a mode. Flags to implement include:

- `-p` or `--parseable`: select parseable mode, over human-friendly formatting
- `-s` or `--separator`: select the separator character
- `-o` or similar: the list of fields to include in the output. It's also nice to print a list of the _available_ fields, in lieu of a man page.
- `-H` or `--no-header`: do not print headers for the selected columns

rmustacc, 2023-06-21:
Note you should ensure that `-p` requires you to pick `-o`. However, `-o` can work without `-p`.

augustuswm, 2025-09-22:
@bnaecker @rmustacc We are using `xcvrdam` in fiber test and output parsing is certainly one of my concerns for it. I'm looking at `ipadm` as an example to model it off of. Does this track?

bnaecker, 2025-09-22:
@augustuswm Yes, that's a good example.

bnaecker, 2026-10-30:
This was implemented in https://github.com/oxidecomputer/transceiver-control/pull/410

closed 2026-10-30 by bnaecker