oxidecomputer/transceiver-control #109: Need a way to acknowledge when Hubris disables a module

Aaron-Hartwig, 2023-06-26:
When something goes awry during the SPs thermal monitoring of a module it will take the following action:
* Power down the module
* Hold it in reset
* Deassert LPMode (to account for the 3.3V leakage when this is driven to some unpowered modules)
* Mark in Hubris that this port has been disabled by policy

That last bullet point has the following effects:
* All messages from the host to change this module will be rejected (need a new `HwError` variant for this)
* Need a new `HostRequest` message to clear the policy

cc: @mkeeter

bnaecker, 2025-02-07:
@Aaron-Hartwig I'm pretty sure we have this now, as part of your work adding the `DisabledBySp` error variant. Is that right?

Aaron-Hartwig, 2025-02-07:
Yes! And the `ClearDisableLatch` message variant.

closed 2025-02-07 by Aaron-Hartwig