oxidecomputer/transceiver-control #127: Want support for decoding datapath information

bnaecker, 2023-07-20:
During bringup of our first customer rack, we looked in detail at an array of transceiver state related to the datapath. It would be helpful to have first-class decoding and `xcvradm` printing of this information. This includes:

- Tx disable bits 
- Tx  LOS / LOL
- Rx LOS / LOL
- Tx / Rx CDR enabled
- Tx fault (adaptive EQ and laser)
- Datapath state (for CMIS modules, not directly supported for SFF-8636)

There is a lot of other information we could gather, but which might be less important:

- Max Tx input equalization
- Max Rx output emphasis and emphasis type
- Host / media side FEC support and enabled state
- Tx squelch, adaptive EQ freeze
- Rx squelch
- Rx output disabled (might be in the first list, but never seen this set)
- TC / CTLE stabilization time

Aaron-Hartwig, 2023-07-24:
@bnaecker I have everything you noted in your second batch of bullet points as part of my [WIP branch](https://github.com/oxidecomputer/transceiver-control/pull/128), so no need to worry about that on your end. We should sync about how we want to expose all this info as I think we are looking at very adjacent things

bnaecker, 2025-02-07:
I'm marking this closed by #300. We can open new issues if some of the more fine-grained details are still missing, but the core datapath state is finished.

closed 2025-02-07 by bnaecker