oxidecomputer/transceiver-control #279: `xcvradm`: want ability to hide modules from upstack software

Aaron-Hartwig, 2024-09-20:
We can find ourselves in the setting where we use the `xcvradm` binary from a Scrimlet's switch zone to debug. However, the very fact that a switch zone exists usually means `dendrite` is running and automatically handing control of transceivers over to the SDE. It would be helpful to be able to gracefully hide modules from `dendrite` using `xcvradm`.

bnaecker, 2024-09-20:
We can currently do this from the switch zone using swadm, using the set-management-mode sub command. Is that sufficient?

Aaron-Hartwig, 2024-09-20:
Oh! Yeah probably 😅

Aaron-Hartwig, 2024-09-20:
@bnaecker I attempted `set-management-mode qsfpN manual` and was still seeing messages come in. It might be worth revalidating that is working?

bnaecker, 2024-09-21:
Yeah, I can definitely take a look. All the command really does is report that the module is not present to the SDE, regardless of what the present pin says. So it's definitely possible for the module to continue to operate in some ways after hiding it. What messages did you see?