oxidecomputer/transceiver-control #299: `xcvradm` should have a command for `ExtendedStatus` messages

Aaron-Hartwig, 2024-12-06:
See https://github.com/oxidecomputer/transceiver-control/blob/3bd33a347ba930e858e64dded1302a13f8e4b34d/messages/src/message.rs#L459

bnaecker, 2024-12-06:
I can't quite remember the history here, but that's actually what we're doing:

https://github.com/oxidecomputer/transceiver-control/blob/3bd33a347ba930e858e64dded1302a13f8e4b34d/controller/src/bin/xcvradm.rs#L698

I recall that @mkeeter and I put this in place with the idea being that we'd start by asking for the extended status, and fall back to the original status message if that failed. But that might have only been during the time when we rolled the new message out -- it's possible that we don't need the status message at all anymore.

Aaron-Hartwig, 2024-12-11:
You're right! I'd completely missed (forgotten, more likely) the fact that we'd extended `status` to have all the `with`, `without`, `all` stuff. My bad!

closed 2024-12-11 by Aaron-Hartwig