oxidecomputer/transceiver-control #339: Not correctly decoding temperature for pre SFF-8636 rev 2.8 modules

bnaecker, 2025-02-22:
Modules report a bunch of diagnostic monitoring data. One important one is temperature. This is required for all separable modules (all the ones we use are), but we recently found that `xcvradm` was reporting temperature as unsupported for a supposedly compliant module:

```
root@oxz_switch0:~# xcvradm -i sidecar0 -t 18 vendor-info
Port Identifier               Vendor           Part             Rev  Serial           Mfg date
  18 Qsfp28 (0x11)            Intel Corp       SPTSBP3CLCCO     01   CRCL203602ZCS    20090500
root@oxz_switch0:~# xcvradm -i sidecar0 -t 18 monitors
Port 18
         Temperature (C): --
      Supply voltage (V): --
       Avg Rx power (mW): [1.0438,1.4170,1.1306,1.0131]
            Tx bias (mA): [47.8800,47.8800,47.8800,47.8800]
           Tx power (mW): [0.8185,0.7188,0.8847,0.9508]
                   Aux 1: --
                   Aux 2: --
                   Aux 3: --
```

The support for the temperature monitor is advertised in upper page 0, byte 220, bit 5. Bit 4 advertises voltage support. In this module, both of these appear "unsupported":

```
root@oxz_switch1:~# xcvradm -i sidecar0 -t0 read-upper --sff --page 0 220 1 --binary
Port Data
   0 [0b00001100]
```

But if you read the actual registers, they sure appear to be reporting reasonable values:

```
root@oxz_switch1:~# xcvradm -i sidecar0 -t0 read-lower --sff 22 2
Port Data
   0 [0x27,0x54]
root@oxz_switch1:~# xcvradm -i sidecar0 -t0 read-lower --sff 22 2
Port Data
   0 [0x27,0x3f]
```

That shows the LSB moving around a bit, which is consistent with small (sub 1 degree) fluctuations. So what gives?

Well, if we read the spec more carefully than I did at first, it says:

> Supply voltage monitoring implemented (0b=Not implemented or pre-Rev 2.8, 1b=Implemented)

So, either temperature isn't supported, or the module is an older rev of the SFF-8636 spec. What rev is it?

```
root@oxz_switch1:~# xcvradm -i sidecar0 -t0 read-lower --sff 1 1
Port Data
   0 [0x07]
```

Which is: `SFF-8636 Rev 2.5, 2.6 and 2.7`.

Well, that explains it. We're dealing with an older module where I'm guessing these bits were reserved, and now they mean something.

One solution here is to either read the status byte, and then use that to interpret the advertisement bits. But a simpler and more robust thing would be to just read the expected bytes for both temperature and voltage, and then report them if they are something "reasonable", where reasonable probably just means not `0x00` or `0xffff`.

rmustacc, 2025-02-22:
I think we should generally be reading and knowing the status byte in order to interpret other fields. It may not work as well depending on how things are factored, but if we aren't using that information later on for interpreting reads and writes that seems like something we'll want to. If we are at >= 2.8 and the device accurately reports that the sensor isn't implemented, then I wouldn't actually do the read and try to interpret it given we've been told it's bogus explicitly versus having to figure it out.

bnaecker, 2025-02-22:
Thanks @rmustacc, good suggestion. I've added a commit to #341 to read the status byte and use that to determine whether we respect or ignore the advertisement bits.

commit eb1b83a2 (2025-02-24): Always attempt to read module temperature and voltage (#341)

* Always attempt to read module temperature and voltage

Fixes #339

* Actually decode status, and use that to determine if the advertisement
bits mean anything.

Add a test specifically for this case, where the module is old but
doesn't advertise support. We should ignore that lack of advertisement,
and decode the value anyway.

* Add reference to status byte table
closed 2025-02-24 by bnaecker